OpenSourceEBike/Bafang_M500_M600 #16: Reorganizing Hardware section?

Tomblarom, 2021-08-01:
Having a folder, called img is very generic. I have pictures of a M500 and M600 controller version and wanted to commit them, but I won't just throw it into that folder. I'd suggest adapting the structure like this:

```
Bafang_M500_M600/
├─ Hardware/
│  ├─ Controller/
│  │  ├─ M600 A18YJV2019911/
│  │  │  ├─ img hardware version
│  │  │  ├─ controller front - annotated
│  │  │  ├─ controller back - annotated
│  │  ├─ M600 A18F8TB089622/
│  │  │  ├─ ...
│  ├─ Drawings/
│  │  ├─ ...
│  ├─ Mechanical/
│  │  ├─ ...
├─ .../
```

In my opinion it's important to trace the hardware changes through the different PCB revisions! We can't assume M500 2018 = M500 2021.

Ornias1993, 2021-08-01:
There are a few important things to note here:
- I was/am already planning to restructure it a bit, though i'm also looking into moving to another documentation platform altogether using github pages, depending on what suit our needs best
- The `img` folder is a store to store images embeded into other content. This is normal development best-practice. We just haven't found the time yet to include them all in documentation.
- It's not wise to have a clusterfuck of folders. No need to seperate revisions unless actually needed and even so, there might only be a few relevant differences.

Ornias1993, 2021-08-01:
Speaking of which... @casainho Could you please enable github pages in settings and point it towards the /docs folder?

casainho, 2021-08-01:
> Speaking of which... @casainho Could you please enable github pages in settings and point it towards the /docs folder?
> I can handle the rest relatively well.

I just gave you maintainer permissions, I hope you can now configure this option.

Ornias1993, 2021-08-01:
> > Speaking of which... @casainho Could you please enable github pages in settings and point it towards the /docs folder?
> > I can handle the rest relatively well.
> 
> I just gave you maintainer permissions, I hope you can now configure this option.

Awesome, thanks!
I'll work on the docs side this month, though I also have some work for another project so it might take a few weeks :(

casainho, 2021-08-01:
Has you or anyone plans to move on firmware development?

Ornias1993, 2021-08-01:
> Has you or anyone plans to move on firmware development?

If anyone could get me the firmware/memory/debug dump, I would (even without hardware) be willing to put some time into figuring out which memory address gets loaded as a firmware entrypoint. Preferably I would reuse the Bafang bootloader if we can, because it would mean users could just load opensource firmware without needing to change the bootloader.

Tomblarom, 2021-08-01:
I'm not that experienced, to help with development, other than testing or providing hardware, like I offer you. Got M500 ready to ship, but currently unsure who shall receive it. As I said, got M500 and M600 for development purposes.

@Ornias1993 can you add them?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Ornias1993, 2021-08-01:
@Tomblarom OMG WOW, those pictures are REALLY good!
I got your email, will answer it tomorrow or the day after... 👍

Ornias1993, 2021-08-01:
THose controller boards are really interesting!
Two m600 boards and one of which has double the FETS and double shuts and the other has not...hmmm...

Tomblarom, 2021-08-01:
I renamed the files to make clear, what is what. You can check again.

**M500 M600 Controller Externals:** Left cables are M500, Right cables and stickers (were stacked, maybe because it was reprogrammed) are M600
**M500 Controller Serial:** There is none, since the controller shell had no serial number on it.

Thought about annotating them, but 4πr^2 (in the ES forums) did an awesome job annotating. Maybe I'll do it later on.

Ornias1993, 2021-08-01:
@Tomblarom There are no file names the way you uploaded them. Also afaik github adds some compression.
Could you send me the raw files?

casainho, 2021-08-01:
@Tomblarom great work! Seems the board is 2 layers only, right? if so, with such good pictures, would be great if someone could draw the schematic even if partially. For instance, to drive the motor, there is need to control the mosfets and read the shunt current, as also battery voltage, I think. Also throttle is needed for input of commanded  motor current. Also UART is needed for output motor variables.

Do you have an oscilloscope to measure some signals??

Tomblarom, 2021-08-02:
I'll try to use my SLR camera and redo the images for easier tracing.

Nope.. Unfortunately got no oscilloscope.. As I said:
> I'm not that experienced, to help with development, other than testing or providing hardware, like I offer you. Got M500 ready to ship, but currently unsure who shall receive it. As I said, got M500 and M600 for development purposes.

Ornias1993, 2021-08-02:
@Tomblarom Could you give a lost of which boards you've got?
Maybe you could send the broken one (unless you destroyed that one for the pictures, which would be a shame really) to me to check if I can power it to dump the firmware. afterwards I could forward it to @casainho relatively cheaply.

casainho, 2021-08-02:
Before sending the boards to anyone, would be great if you could use a multimeter to follow the connections on the board and draw a partial schematic. Do you have time and motivation to do this? If so, I can guide you on how to draw the schematic.

Ornias1993, 2021-08-02:
> Before sending the boards to anyone, would be great if you could use a multimeter to follow the connections on the board and draw a partial schematic. Do you have time and motivation to do this? If so, I can guide you on how to draw the schematic.
> 
> That pictures are great for helping following the connections and do it physically is a must for validation.

Agreed, tracing would be great!

Tomblarom, 2021-08-02:
> Do you have time and motivation to do this? If so, I can guide you on how to draw the schematic.
Unfortunately not. Fairly busy with other projects. That's the best I can do for you right now.

Ornias1993, 2021-08-02:
> > Do you have time and motivation to do this? If so, I can guide you on how to draw the schematic.
> 
> Unfortunately not. Fairly busy with other projects. That's the best I can do for you right now. I added pictures, I took with a SLR camera. Maybe they are better for some parts. Also added the status of these two controllers (functional vs broken).
> 
> @Ornias1993 I only got an RD6018 and basic electronics tools. I can't offer anything else, than providing this in-depth material, since fairly I'm busy...

With equipment I was refering to a list what m500 and m600 hardware you got available to distribute...

Tomblarom, 2021-08-03:
- Broken M500, M620 controller
- New M600 controller
- M500, M600, M620 CAD
- Broken SW102

Tomblarom, 2021-08-03:
Who shall receive my hardware? Can you discuss this?

casainho, 2021-08-03:
On next 2 weeks, If I had the hardware, I would only try to connect the flash/debug JLink to see the boards works on that - I would not be able to do more.

And I really think the next step must be do partially the schematic, as motor controller firmware can not be develop without knowing the connections from the microcontroller to the hardware. It needs someone motivated and with time to do that, using the boards, the pictures and a multimeter.

Ornias1993, 2021-08-04:
@Tomblarom You can send it all to @casainho I think thats the most tactical approach atm.

Tomblarom, 2021-08-05:
@casainho do you want me to send the Higo cables + Innomaker HW with it? I definitely need them + M600 controller (working!) back, at some point.

[image]

casainho, 2021-08-05:
For the development, all that components will most for sure be destroyed.

Tomblarom, 2021-08-05:
I think you can get the most from the broken M500, without destroying the M600. Do it if you really have to, but I'd be happy to get it back. - Can you at least answer the main question, I asked? I even posted a picture??

Tomblarom, 2021-08-06:
As I got no feedback, I only provide the broken M500 and M620, since I have no guarantee to get my freely provided hardware back.

Ornias1993, 2021-08-06:
@casainho you should’ve respondes on this.

casainho, 2021-08-06:
Sorry, I am being riding for some days, away from computer.

If there are special controllers and with special firmwares that are sold for 1000 euros, then I think this project can be a BIG problem for that companies.

I understand that development tools are expensive, but there is not other way. And it is very expensive for one person only donate a complete part like a motor or a bicycle. I think the best could be probably asking donations in money, so everyone could done a small part.
In fact someone did that, collected the money and bought me the TSDZ2 V2 motor controllers and a motor.

About this project, I would really need a motor and an ebike with it, so, I should stop dreaming with this as I have no space at home for more bicycles. Maybe next year? Anyway, with the controller only, I can give a look.

[image]

Ornias1993, 2021-08-07:
Lets keep this on-topic from now on.
Please create new issues for new subjects (like sending hardware and such).

@Tomblarom I'll do some restructuring this week and close this ticket once done.
I think that solves you issues :)

Ornias1993, 2021-08-07:
I've hidden the whole off-topic discussion about sending hardware. As it has nothing to do with the subject of this ticket and is also mostly resolved by now :)

Tomblarom, 2021-08-09:
One last off-topic: I sent out the parcel, recently. ;)

cpardo13, 2021-10-11:
Hello.
I have tested the different firmwares in my 36v m600 motor.
The original firmware is the CRX3625i and have around 900w of peak power.
The firware CRX3615i have around 600 w of peak power.
I tested the experimental firmware 36v-20a and the peak power it's the same as 15a, around 600w.

With the 25a firmware, the power at PAS 1 is very high, around 400w.
With the 15a firmware, the power at PAS 1 is around 250w (better for me than 400w) but the power at PAS 5 is too low in comparison with the 25a firmware, only 600w.

The best woul be a low power at PAS 1 but with the max power at PAS 5 or medium step betwen the 25a and 15a firmware. 

A 20a firmware It would be ideal. 😭😭😭