OpenSourceEBike/Bafang_M500_M600 #25: M500 (M600?) Current limit

KyokushinPL, 2021-08-03:
Its been discovered the 0x774 in firmware is corresponding to the Current Limit, but it seems to be not exactly that, because:
On "CRX10NC4812i142015.5-12A-20210616.bin" what is reporting after flashing as "CRX10NC4810i142015.5.bin i have around 570W in peak.
On "CRX10NC3617i142017.7-28-2230-20200911.bin" is reporting on photos around 750W or more:
https://www.emtbforums.com/community/threads/bafang-m500-m600-tuning-the-besst-tool.18860/page-3#post-323051

However, i have noticed when i changed in CRX10NC4810i142015.5-20191126.bin 0x774 to 0E (14Amps) it bumped up max current to 672W, but when i tried 0F 0A there is no change. There is also no change on 18Amps.

So 0x774 may mean something different than max limit or maybe there is also some multiplier or some other addres also responsible for the limits.

I am opening the Current Limit thread for discussion, to keep it separated.

Ornias1993, 2021-08-04:
Thread is fine with me :)
Thanks for the hard work :)

doctorrevenge, 2021-08-04:
[image]
i thinck somebody at leaderfox has a tool to change firmwares. We need a contact ;)

Ornias1993, 2021-08-04:
@doctorrevenge I got multiple reports about this brand already, I call it a hoax and fake for now. As there are no m500 firmwares on the website. 
Instead of a screenshot, please actually provide a download URL.

KyokushinPL, 2021-08-04:
For now the most mysterious (at least for me) files are:
CRX10NC4812i142015.5-12A-20210616.bin (0A on 0x774) - stronger than 10 Amps in real - around 570 (or higher, need re-test)
CRX10NC3617i142017.7-28-2230-20200911.bin (0F on 0x774) - stronger than 15Amps, possibly around 750W
because they have measured and confirmed significantly higher 'peak' power than Hex value on 0x774 is stored

doctorrevenge, 2021-08-05:
https://www.leaderfox.cz/en/guides

You Can download it there. Scroll to the end. I tested the Firmware for m300 and m500. Both works

Ornias1993, 2021-08-05:
> https://www.leaderfox.cz/en/guides
> 
> You Can download it there. Scroll to the end. I tested the Firmware for m300 and m500. Both works

I just get a 404

Ornias1993, 2021-08-05:
@doctorrevenge can you send everything and screenshot by enail possibly?

KyokushinPL, 2021-08-05:
> 
> 
> @doctorrevenge can you send everything and screenshot by enail possibly?

https://www.leaderfox.cz/data/file/BESST%20Bafang%20mid%20motor%20%20firmware%20AJ.zip here is direct url

cpardo13, 2021-08-05:
@KyokushinPL hello.
The m600 36v25A firmware also have 0F in 0x0774. With around 900w of power.

doctorrevenge, 2021-08-05:
Maybe we must try to make it possible to Flash m600 36V 25A in a m500 Motor and See whats happening. Maybe it is easier to change m500/600 ID in Firmware to make it flashable...

Ornias1993, 2021-08-05:
@doctorrevenge Got your firmwares and added them to the catalog :)

Great idea btw, i'll create an experimental firmware to try flash m600 on m500 boards 👍

Ornias1993, 2021-08-05:
I've just added 2 firmwares (one for 36v and one for 48v) that are m600 firmwares with a change that should force BESST to flash them to m500.

KyokushinPL, 2021-08-05:
Its working and i seen max 990 W output so it seems to be 19-20Amps limit. However, i reverted to M500 firmware. M600 have slightly different gears, different assistance and finally less MOSFETs on controller board, i would like to avoid burn controller. 
Engine should handle that, finally its same as M600 except one plastic gear, but i am not sure of the controler will  handle 1kW in daily usage.

Ornias1993, 2021-08-05:
> Its working and i seen max 990 W output on my 48V M500 so it seems to be 20Amps limit. However, i reverted to M500 firmware. M600 have slightly different gears, different assistance levels probably and finally less MOSFETs on controller board, i would like to avoid burn controller.

Thats interesting, thanks for your feedback... Thats some success right there :)

> Engine should handle that, finally its same as M600 except one plastic gear (early M600 models had same plastic gear), but i am not sure of the controler will handle 1kW in daily usage.

Yeah the actuall Electromotor shouldn't be the issue here.
That being said:
The M600 with double MOSFETS, should be able to handle more than 2KW, so I think 1KW with half the MOSFETS should work out fine 👍

Ornias1993, 2021-08-05:
I'll port all m600 firmwares with the same patch, so we can try that out... Should be a few with lower power limits as well :)

KyokushinPL, 2021-08-05:
> The M600 with double MOSFETS, should be able to handle more than 2KW, so I think 1KW with half the MOSFETS should work out fine 👍

how about shunt resistors? M600 have two, M500 one.

Ornias1993, 2021-08-05:
> > The M600 with double MOSFETS, should be able to handle more than 2KW, so I think 1KW with half the MOSFETS should work out fine 👍
> 
> how about shunt resistors? M600 have two, M500 one.

Shunt resistors and amp-ratings really aren't the limiting factor here. One shuntresistor isn't magically going to blow up the board, simply put: MOSFETS have WAY lower durability because they need to handle all the powerflow, shuntresistors do not have to hence can handle way more power.

In theory halfing the amount of resistance should half the amp readings. Which it clearly didn't do in your test...
I expected for a while that the firmware might be able to handle both single AND dual resistors or that the resistors in the m600 combined have the same rating as the single resistor in the m500. Which could explain why the firmware worked for you at all :)

KyokushinPL, 2021-08-05:
I have observed the temperature of controller and engine and there was no noticeable (in my not reliable test) higher than on M500 f/w.
I just put highest gear, hold the rear breake (to simulate the road resistance) and push up the throttle to achieve highest possible value. Its silly way of testing at home but if something is working then its not silly ;)

KyokushinPL, 2021-08-06:
I have performed additional tests, as i suspected the stock f/w "CRX10NC4812i142015.5-12A-20210616.bin" have 672W peak otuput what is meaning this have 14Amps not 12 Amps.

It seems 0x774 does not mean max current, however in my opinion its connected with current somehow but with rather with degradation of power in battery charge function - for example - on 50% of battery you will have more peak power on 0F than on 0A. I need to make more road tests for this.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Andrzej/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Andrzej/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Bafang firmware peak power output |   |   |   |   |  
-- | -- | -- | -- | -- | --
  |   |   |   |   |   |  
no | filename | MD5 | 0x774 value | measured peak power | engine model | other
1 | CRX10NC4810i132003.0-28-2230-20210114-2.bin | 2ba44bd1fc96f7305d8f386e5dcd4fae | 12 | 576 | M600 |  
2 | CRX10NC4812i142015.5-12A-20210616.bin | 8f37da1b65ae85d87938fc91b495f0f | 0A | 672 | M500 |  
3 | CRX10NC4818i132014.6-20191126.bin | b79263073a6d7a9005c405b7e66603f2 | 12 | 1000 | M600 |  
4 | CRX10NC4818i132046.6-28-2230-20200824-3.bin | 4916960cce34b0e3e60ab9d1502ccfb9 | 12 | 1000 | M600 |  
5 | CRX10NC4818i132046.7-28-2230-20201121.bin | 7d0de61112c47456556f5520ab679154 | 12 | 1000 | M600 |  
6 | CRX10NC4810i142050.6-22-1768-20201109.bin | 3b5c8c55adec18664418f7db1b7ecea7 | 0A | N/A | M500 | throttle locked  to walk mode low speed (around 5km/h),   unable to test

</body>

</html>

doctorrevenge, 2021-08-06:
i have tested the modded m500 36V 25A Firmware... it works. i see peaks of 1000W and 25A but the 36V17A is more aggressive... I like it more. a dream is the peak from 36V25A in the 36V17A firmware ;)

KyokushinPL, 2021-08-06:
what peak you have on 36V17A ?
what exact filename you tested for 25A?

doctorrevenge, 2021-08-06:
On 17A ca. 750w an 20A peak
On "625" 25A testfirmware ca. 1000w an 25A peak
But the 17A firmware is better... maybe my Battery or bms cant handle 25A stable?!? What about the guy with the crussis largo 9.6 he has the biggest battery from crussis installed... maybe it works better on his bike.

KyokushinPL, 2021-08-06:
battery surely is fine, its same one as for M600.
Maybe its not 1000W, maybe its only displaying 1000W, its worth checking. Vmax test on full throttle should verify that.
If peak is a true 1000W then vmax will be higher no matter what.

Ornias1993, 2021-08-06:
> i have tested the modded m500 36V 25A Firmware... it works. i see peaks of 1000W and 25A but the 36V17A is more aggressive... I like it more. a dream is the peak from 36V25A in the 36V17A firmware ;)

You mean the firmware starting with 625?
Please people actually use filenames like @KyokushinPL

Ornias1993, 2021-08-06:
@KyokushinPL youre right, it might display more than is actually used due to having one less resistor!

vMax? Velocity max? Max velocity and max rpm wont change with current afaik...

KyokushinPL, 2021-08-06:
If peak power is current then max velocity (vmax) will be higher, same bike, same gears, more power = higher speed on WOT (wide open throttle)

Ornias1993, 2021-08-06:
Ahh vmax under zero resistance? Yeah that should be different indeed :)
Though you would also need to nuke the speedlimits and such.

KyokushinPL, 2021-08-06:
The maximum speed on road test, on same road and same resistance. On zero resistance he will hig 60km/h limit always.
We need a simulation of dyno run in same and reatable test environment.

So if @doctorrevenge need to go to flat road, put max assistance and ride so fast as he can on only throttle at the end (to remove the human assistance factor from measurment).
And IF this 1000W is real then he will achieve higher maximum speed.

KyokushinPL, 2021-08-06:
@Ornias1993 
the files 
CRX10NC4810i142015.5-20191126.bin and 
CRX10NC4818i132014.6-20191126.bin
looks similar, same date, one is m500 and one m600, one have 570 peak and second around 1kW.

Ornias1993, 2021-08-06:
So? What are you responding on?

KyokushinPL, 2021-08-06:
I tried to compare them in matter for "max current" difference, something other than 0x774, but dont have clue how to move it forward :/ maybe these two be somehow useful to you.

Ornias1993, 2021-08-06:
@KyokushinPL i suggest starting to focus on 625 vs normal m500 firmwares instead.

KyokushinPL, 2021-08-07:
I done the road test and:

Adapted M600 f/w **(625-CRX10NC4818i132046.7-28-2230-20201121.bin)** is showing 1kW in peak but bike is noticeable slower, i hit 35km/h on throttle
Stock M500 f/w CRX10NC4812i142015.5-12A-20210616.bin with 670W in peak is hitting 40km/h on throttle.
- below 30% the bike is cutting the power to 10 Amps (no matter of 0x774 value)
- i have been compared stock CRX10NC4812i142015.5-12A-20210616.bin and Experimental firmware 48v18a.bin on low-power circumstances (battery drained to around 40%) and it seems giving same output powre, so 0x774 change seems to changing nothing. My first findings was misleading because i had 'out the box' the firmware named CRX10NC4810i142015.5-20191126.bin what is reporting "CRX10NC4810i142015.5" same as "CRX10NC4812i142015.5-12A-20210616.bin" which is noticeable (100W) more powerfull one.

So it seams we back to the beginning.

Ornias1993, 2021-08-07:
@KyokushinPL Well not really. 
- you've proven that m600 firmware can actually be loaded on m500 boards
- You showed that the m500 boards indeed have half the resistors and thus a wrong resistance reading when loading m600 firmware
- You've hence created an interesting case in that we could load m600 firmware with full functionality by adding the missing resistor and loading m600 firmware

**Anyway:**
Could you make a list/table for each firmware for the m500 (official firmware that is) what your max wattage reading is? I could add that info to the readme to help people select firmware based on actual performance instead of filename!

KyokushinPL, 2021-08-07:
A list with wattage is done (for original Bafang firmware and 48V version), you need to scroll table right.

I had also thinking about the adding the missing hardware (mosfets and resistor) will allow to achieve real peak 1kW output and turn M500 board to M600 board or not, however i would be more happy to add +100W to M500 peak and adjust this current decay threshold on lower battery voltage - its annoying.

Ornias1993, 2021-08-07:
> @Ornias1993 A list with wattage is done (for original Bafang firmware and 48V version), you need to scroll table right.

I want a clear table with ALL m500 firmwares.
I'm seeing a few m500 and a few m600 firmwares.
Can you make something more clear please, also: donnot modify the firmwares. Just the official firmwares.

> I had also thinking about the adding the missing hardware (mosfets and resistor) will allow to achieve real peak 1kW output and turn M500 board to M600 board or not, however i would be more happy to [...]

No you do NOT need the extra mosfets for 1KW output. just the resistor. 
I'll explain again: M600 mosfets are build for 2kw peak. m500 for 1kw peak.

> I do not know how people are using they're ebike with smaller battery, because for me 840Wh is not enough (around 55km range on full 5 assist/throttle)

They do not use full 5 assist and no throttle (because that is even illegal in the EU)

KyokushinPL, 2021-08-07:
Table for M500 48V (only known official):

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Andrzej/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Andrzej/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

filename | measured peak power | voltage | reports after flashing | comments
-- | -- | -- | -- | --
CRX10NC4812i142015.5-12A-20210616.bin | ~672 | 48 | CRX10NC4810i142015.5 | reporting same name as   CRX10NC4810i142015.5-20191126.bin
CRX10NC4810i142015.5-20191126.bin | ~576 | 48 | CRX10NC4810i142015.5 |  
CRX10NC4810i142050.6-22-1768-20201109.bin | N/A | 48 | CRX10NC4810i142050.6 | throttle locked  to walk mode low speed (around 5km/h),   unable to test
</body>
</html>

If i will add the resitor and get all of it opened and dismounted then i also can add the mosfets, its same work and will be more reliable.

In my country (Poland) the case of throttle is not fully clear (it depend who is interpreting the law) - a grey area (it may be detached anytime), the law is saying about the motor, but nothing it can have throttle or must not.

However there is excellent BOOST function what is giving max output after hitting the 25km/h limit what is working like on/off throttle in fact. There is one firmware for 36V version with lowered this limit to 15km/h, i have even try to find in what hex it can be stored but found nothing.

doctorrevenge, 2021-08-07:
On the CRX10NC3617i142017.7-28-2230-20200911.bin
Max speed is ca. 55kmh on Bicycle stand

On the 625-CRX10NC3625i132047.7-28-2230-20201121.bin
Max speed is ca. 45kmh on Bicycle stand

With the 17A firmware the bike Shows most the time 750W till Vmax.
With the 25A firmware the bike Shows ca. 1000W but not often. It "jumps" between 500 - 1000 but Not constant... maybe the Controller from m500 is not strong enough for the 1KW?!?! Or 25A? Anybody test the 48V M600 with the M500 Motor on the road? Not only on Bicycle stand? What about the Vmax?

doctorrevenge, 2021-08-07:
[image]

doctorrevenge, 2021-08-07:
With 17A Firmware on flat road...

doctorrevenge, 2021-08-07:
Maybe we must find the difference between the Stock 15A 36V m500 firmware and the 17A m500 firmware. The 17A firmware was sent to me by an bafang engineer... so tuning file is direct from Bafang. No Mods ;) and the difference to the "standard" firmware is amazing... for the 48V m500 Motor it doesnt exist tuning firmware from Bafang?! I asked for it, but the bafang employer wanted a serial Nr from the Motor, i cant answer ;) maybe you can sent me a number and i ask for it again.

Ornias1993, 2021-08-07:
@KyokushinPL Awesome table, thats what I wanted :)
mind finishing the power readings for all of them? That way we can add that the completed table to the readme...

> If i will add the resitor and get all of it opened and dismounted then i also can add the mosfets, its same work and will be more reliable.

You could. But for a novice solderer, the shunts are easier to solder because they have bigger pads, that was what I was refering to ;-)

> In my country (Poland) the case of throttle is not fully clear (it depend who is interpreting the law) - a grey area (it may be detached anytime), the law is saying about the motor, but nothing about bike can have throttle or must not.

Nope, EU regulation is clear as day. Just because Poland did a bad translation is kinda irrelevant. Said EU law is a regulation and not a directive. Regulations have direct power of law in all member states. To be precise: `no. 168/2013 art 2 sub 2 under h` prohibits throttles, as power is required to be cut when you stop paddling.

@doctorrevenge 

>  maybe the Controller from m500 is not strong enough for the 1KW?!?! Or 25A? Anybody test the 48V M600 Firmware with the M500 Motor on the road? Not only on Bicycle stand? What about the Vmax?

PLEASE stop. I already know why the performance was lower. It is a very normal response due to having half the resistors. Its 100% what I initially expected when I ported m600 firmware to m500. There is nothing to worry about and 100% understood. It has nothing to do with the controller not being strong enough. It's all 100% clear.

KyokushinPL, 2021-08-07:
@Ornias1993 in Poland our regional law is over anything else. UE wanted to change that not so far ago, but our gov don't agree, so if our law will tell us to wear only red and white then we have to, it does not matter what ue say ;)
As I wrote, we have different categorisation than rest of EU, we have no speed pedelec, ebike, L1A etc.
We have bicycle (powered up to 25km/h, nothing about throttle, but should be cut to 25km/h) <250W, motored-bicycle (250<4kW), motorcycle 4kW, and only bicycle can use bike road lanes and only bicycle don't require driving licence. I have full A motorcycle driving licence so i can ride anything.
So if anyone want something more powerful or faster its simply better to buy a legit motorcycle (and i had two cruisers). At now i am going to use bike lanes and mostly i am riding <25km/h.

Unfortunately i have 48V motor so i cant fill the 36V wattage, i have filled that for all 48V including that M600 adapted ones.

Ornias1993, 2021-08-07:
> @Ornias1993 
> I understand clearly what you wrote, but PL is specific case, please belive me.

No, you're wrong. 
I think you do not have any formal legal education? (and with that I'm not talking about a single course within some other form of education) Because you clearly have some issues with some basic principles in international and european law.

In those other legal maters (between the EU and Poland), it was indeed the case that the polish constitutional court states that the polish constitution takes precedence over EU law. Which is actually not that different from Germany for example. Actually: the same discussion (but with less publicity) has been ongoing with germany since the start of the EU.

However: What I explained is not related to those current ongoing cases. EU regulations, even in Poland, are directly applicable. The Polish Consitutional court has not decided otherwise either.

> As I wrote, we have different categorisation than rest of EU, we have no speed pedelec, ebike, L1A etc.

Speed pedelecs are L1B ;-)
But: you actually do have L1A and L1B, Poland has no freedom not to accept those definitions.
However: EU countries are free not allowing L1A and L1B to be driven on public roads ;-)

The EU does not dictate how a certain catagory can be ridden in a country, they just define the catagories. It is a very hard and complication difference, but it has to do with the fact that the EU can make regulations about goods (such as defining good catagories) but cannot define traffic rules (including weither or not a vehicle can be ridden in a certain country)

> We have bicycle (powered up to 25km/h, nothing about throttle, but should be cut to 25km/h) <250W, 

You can already see they tried to implement precisely that regulation, but just missed some notes when applying them. So they actually do try to follow EU classifications

> motored-bicycle (250<4kW)

Poland has chosen to combine L1B and L1E, which is actually the same as the Netherlands has done in practice for example. So Yes: They do have speed pedelecs. Keep in mind: The throttle requirement does not exist for L1B and L1E ;-)

> motorcycle 4kW, and only bicycle can use bike road lanes and only bicycle don't require driving licence. I have full A motorcycle driving licence so i can ride anything. 

These are traffic rules and are not defined by EU law (see above). So yes: These can be different from other EU countries. Except one thing: ebikes <25kmh <250w without throttle cannot be classified something other than a bicycle by memberstates. Which means they cannot add a license requirement for them.

> Literally every ebike here have no limit and mostly are around 1kW (people don't care), however it will change i expect soon so i wanted to have EU legit 250W bike with possibilty to remove and add restrictions, detach throttle and have legit wattage on engine plate and serial numbers, and honestly this M500 seems to be perfect (even on 670W, but 36V with 750W seems to be better choice, but i was not eware of it (probably no one was) )

I agree that the m500 is a pretty safe bet when it comes to EU compliance, yes.

> Unfortunately i have 48V motor so i cant fill the 36V wattage, i have filled that for all 48V including that M600 adapted ones.

Ahh sorry that explains things, yes :)

> I am little curious why my vmax on 1000W M600 f/w is 10km/h lower than yours. Colud you write what teeth you have in bike and what tires (i have 27.5x2.8)?

Lets not get offtopic please. It's not relevant at all, because these tests where JUST designed to see if there is ANY difference. There are, byond that it's not relevant because I already know with 100% certainty what caused it. 
Those other metrics are just the tests being inaccurate.

So please do not derail the topic at hand with irrelevant additional information which only confuses people.

> I am trying to compare these different f/ws and i have no idea how to dig where this max current is stored :(

Donnot even try, it takes specialist knowhow to do this.  There are no easily defined current definitions byond what I already documented in terms of possibilities.

> Maybe the value is for one transistor and M600 it just doubling, so maybe the the 570W M600 will have a half of power of M500 570W, maybe that comparison is worth checking - its only my speculation

Please stop guessing here, for everything with electro technical knowhow it's super clear what happens here. 

Half the resistence on the shunt resistor (aka: 2 instead of 1 shunt resistor) means it reads half the current, double the resistance means it reads double the current.

So the m500 just reads double the current of the m600. That was what I already expected and why I didn't create those 625 firmwares initially.

Ornias1993, 2021-08-07:
> No, you're wrong.
> I think you do not have any formal legal education? (and with that I'm not talking about a single course within some other form of education) Because you clearly have some issues [...]

> please kindly take a look on
> CRX10NC4810i132003.0-28-2230-20210114-2.bin (570W) )vs
> CRX10NC4818i132046.6-28-2230-20200824-3.bin (1kW)
> they're almost identical

Can you please not tell me how to do the reverse enginering steps I already have done to get you these experimental firmwares and documentation in the first place? >.<

Really, I understand you want things to work out...
But you are going to explain basic EU legal issues to someone with a legal background, guessing how a shunt resistor works to someone who know how a shunt resistor works and explaining how to reverse engineer firmware values to someone who just wrote the documentation on it. :')

KyokushinPL, 2021-08-07:
Only one thing what is  obligated PL citizen to know for riding a car or bike is our road law and that one is defining a bike here, and here and only here are bicycle is defined, and as i wrote, there is nothing about throttles and internal discussion is about how it should be understood - about case its not defined or its implied from EU as you wrote (law guys are arguing about that). I am not saying you're wrong - just explaining how it looks. 

However, M500 is EU compliant and have BOOST mode what is pushing engine to mode where is exceeding 25km/h and paddling is not required.

I have no studied international law, but computer science.

Ornias1993, 2021-08-07:
> about case its not defined or its implied from EU as you wrote.

It's literally stated in EU law. There is no legal debate possible about that. None.
Sorry, but you're wrong here.

> I am not saying you're wrong - just explaining how it looks. I have no studied international law, but computer science :) Law is not my area.

Maybe you should defer to people that did in that case, instead of repeating yourself when I literally pointed you towards the regulation that prohibits throttle AND is directly applicable in memberstates due to the legal weight of EU regulations on the definition of goods within the internal market.

> However, M500 is EU compliant and have BOOST mode what is pushing engine to mode where is exceeding 25km/h and paddling is not required.

Boost is only allowed while paddling below 25km/h for electric bicycles in the EU. period.

For reference (because you clearly didn't look it up yourself):

> pedal cycles with pedal assistance which are equipped with 
> an auxiliary electric motor having a maximum continuous 
> rated power of less than or equal to 250 W, where the 
> output of the motor is cut off when the cyclist stops 
> pedalling and is otherwise progressively reduced and 
> finally cut off before the vehicle speed reaches 25 km/h;

**Modbreak**
When it comes to the legality of throttles, I'm actually going to moderate on disinformation from now on.
The EU regulation and priority of law in the EU when it comes to the internal market is super clear.

According to LEVA-EU and the European commission It's a crappy regulation and there are ongoing talks to fix the many fuckups in said regulation. It but it is actually super clear.
(the crappy part is the fact it creates needless problems, because both heavy and light bikes are all limited to 250W, which is unworkable.)

Ornias1993, 2021-08-07:
> I am also not trying to tell you how to do rev eng, I am far away from that - i have a hope to learn something from you - i have totally no experience with reversing hardware or its software. I really do appreciate your work here and big thanx for that, its amazing for me what is happening here where i am reading new updates to proejct.

Finding these values is what is called reverse enginering.

> As i remember for 'almost identical' comparison was used other firmwares whre 0x774 was discovered, but these one what i wrote here just above have identical same first blocks, identical middle, end and, even 0x774, different is only name and **literally few bits**.

I've compared a lot of firmwares, you are assuming a lot.
I've spend DAYS and DAYS on end working on these firmwares.

I only posted about 5-10% of what I did in terms of research, because I only send the highlights.

KyokushinPL, 2021-08-07:
Ornias, you pointed the regulation but i am telling you the PL citizen may totally dont care about this regulation because important for him is only PL road law and their definition of bicycle. Art 2.47 PoRD and that is the very only story what is interesting the bike rider.

Ornias1993, 2021-08-07:
> Ornias, you pointed the regulation but i am telling you the PL citizen may totally dont care about this regulation because important for him is only PL road law and their definition of bicycle. Art 2.47 PoRD and that is the very only story what is interesting the bike rider.

Sorry, but you're wrong here. Polish law is not the only law that applies to polish citizens (or any citizens within the EU single market). I know it's very complicated, but it's not disputed by any legal scholar that EU market regulations have direct application within the EU single market.

KyokushinPL, 2021-08-07:
Different thing is to allow something to sell in EU and here a bike need to be compliant with EN 15194 and different story is a road law how the bicycle defined is. And these stories are different.

Ornias1993, 2021-08-07:
No, it's not "legal" or "illegal". It's either a bike or "not a bike".
It requires type-certification or doesn't.

Stop talking about 15194 btw, thats not related to the subject at all.

"REGULATION (EU) No 168/2013 OF THE EUROPEAN PARLIAMENT AND OF THE COUNCIL"
Enforces type-registration for all vehicles that are relevant to us.
It exclude some vehicles and is 100% clear that throttles mean a vehicle requires type-certification.

If Poland is going to somehow type-certify ebikes with throttle and keep it seperate from other speedpedelecs within the same vehicle class... they can theoretically allow throttle. but NOT without type certification.

Ornias1993, 2021-08-07:
But those options aren't relevant in this case. There is no other catagory of vehicles in Poland.

Poland fucked up and made a mistake codifying EU law into national law. Which happens quite often.
Nothing more nothing less.

KyokushinPL, 2021-08-07:
there is no speed pedelec in Poland :) As i told you, there is motorcycle (and that is defined according to EU types), motored-bike and bike and i gave you a source of these definitions (and only existing ones in poland).

Ornias1993, 2021-08-07:
> there is no speed pedelec in Poland :) As i told you, there is motorcycle (and that is defined according to EU types), motored-bike and bike and i gave you a source of these definitions (and only existing ones in poland).

DUDE.
I explained to you already that just because it isn't CALLED a speed pedelec in Poland, doesn't mean it doesn't exist.
You really have no idea what you are talking about, have no legal background and keep trying to educate me without understanding the subtle nuances of the law.

KyokushinPL, 2021-08-07:
I gave you law article with definitions and only definitions in Poland which should care a person who is driving a bike.

Ornias1993, 2021-08-07:
One more time:
1.
Those LnX EU catagories automatically exist in all EU countries.
HOWEVER: They donnot all need to allow them on the road.

2.
In law it doesn't mater if it's called a "speed pedelec".
It just maters if it fits within a description. L1B and L2E fits perfectly fine within polish law you described. :-)

KyokushinPL, 2021-08-07:
Categories exists but they're are not implied to the road law, and in road law there is no LnX categories. For example, the motorcycle is based on LnX (you may see even references) but bicycle or motored bike not, and that is mostly why definitions from EU regulations have no road law implications and as you noticed there is difference with codyfing pl law and eu. Intentional or not, but it is.

Okay - i tried to show you how it looks from Polish side but you're started to delete the comments even you got the Law article you asked for. 

Over. Good night :)

KyokushinPL, 2021-08-07:
Categories exists in EU but they're are not implied to the Polish road law, and in road law there is no LnX categories. For example, the motorcycle is based on LnX (you may see even references) but bicycle or motored bike not, and that is mostly why definitions from EU regulations have no road law implications and as you noticed there is difference with codyfing pl law and eu. Intentional or not, but it is.

Okay - i tried to show you how it looks from Polish side but you're started to delete the comments even you got the Law article you asked for.

Over. Good night :)

Ornias1993, 2021-08-07:
**Anyhow:**
Totally unrelated, but some topics have become totally unreadable so I'm going to close topics after about 50 posts from now on.
So I'm closing this, to prevent this from becomming unreadable.

Summary:
- our experimental firmwares did not work
- Uploading m600 firmwares to the m500 does work
- m600 firmware on an m500 has issues with the fact the m500 has half the shunt resistors so it reads more current (double) than actually goes through the motor
- Polish law has compliance issues with EU reg no. 168/2013

closed 2021-08-07 by Ornias1993