OpenSourceEBike/Bafang_M500_M600 #37: Building on Luna's work

Ornias1993, 2021-12-22:
As Luna's Ludi v2 work is opensource but their hardware is not...
Maybe we can actually work on the hardware now that we have opensource software?

VESC itself uses an industry standard STM32 variant and have schematics available online:
https://vesc-project.com/sites/default/files/Benjamin%20Posts/vesc_75_300.pdf

If we can get a board diagram of the ludi v2 that would be even easier!

@casainho @CiDi-IT and thought on this?

casainho, 2021-12-22:
I am working with other Xiaomi M365 developers to develop an OpenSource EBike and EScooter motor controller. My idea is to have a generic  EBike and EScooter motor controller, with room for custom electronics, easy do DIY and repair.
I will use it first on my Xiaomi M365 electric scooter and then probably move to use it on EBikes motors.

My idea is to have room for some custom electronics, so I can reuse it over different projects / devices. Both firmware and electrics will be reused, so this means future prof and independent of manufacturer. Sure, specifics customs electronics will need to be developed, like the specifics to read the torque sensor of Bafang M800.

It will take time, but I will get there over time.

[image]

Ornias1993, 2021-12-22:
Is there a reason to keep it not VESC-project compatible?

casainho, 2021-12-22:
In terms of hardware, VESC 4 only supports 12S battery. It is hard to repair. Has no room for custom electronics / customizations.

As for firmware, I found a more simple FOC firmware. May not be the most complete motor controller implementation, but it is easier to understand and most probably easier to port to other different microcontrollers in future, if needed. Also the developer is very active and friendly. Here is: https://github.com/EBiCS/EBiCS_Firmware

Ornias1993, 2021-12-22:
@casainho I actually agree...
I was reading through the VESC sourcecode and it's pretty much a clusterfuck.

That project looks pretty need! :)
Whats the current status of your work, anything opensource so people can actually help with it?

casainho, 2021-12-22:
By the way, there is now SimpleFOC, motor control for Arduino!! https://simplefoc.com/

And it has the advantage to be easy to customize electronics and firmware, since it is an Arduino shields and Arduino firmware!!

But the hardware is low power, without focus on EBikes. Also the firmware is not so good to implement FOC, at least it is what others are saying. Still I really like the results they seem to get. Who knows if in future we can have a shield / motor controller as the one I am doing and run SimpleFOC firmware on EBikes??

[image]

casainho, 2021-12-22:
And about the display, I already have a stable version of DIY, simple and easy to build and repair display. The firmware it is also easy to customize for other devices.

[image]

Simple schematic (for Bafang M800, just need to add the module for CAN):
[image]

So I already have the display, next I am working on the motor controller. Finally the missing parts is the specifics of each device. For instance, Xiaomi M365 scooter uses UART singe wire to communicate with the display and Bafang new motors uses instead CAN. The scooter uses only throttle and M600 uses throttle but also a torque sensor and cadence sensor.

casainho, 2021-12-22:
I am working on this Telegram channel: https://web.telegram.org/z/#-472926861

casainho, 2021-12-22:
Invitation to the telegram channel: https://t.me/+HDBKjVIwfiDAxQD5

Tomblarom, 2022-02-05:
@casainho in fact, this not really related to the aim of this issue nor the repo in general.

casainho, 2022-02-06:
About reusing technology for EBikes motors, we got our OpenSource firmware running an Xiaomi M365 EScooter motor(1) as also running the TSDZ2 mid drive ebike motor(2). This same firmware did run on our OpenSource and DIY motor controller(1) as also on the M365 motor controller -- see the videos bellow.

You guys feel free to reuse this technology on this project!! We are building this as modular, we now have the motor controller hardware (with room for custom electronics), the motor controller firmware, the display (hardware and and firmware). 

- Motor controller with room for custom electronics, easy DIY and repair: https://github.com/EBiCS/EV_Motor_Controller_Customizable_Electronics
- Motor FOC firmware only as a library: https://github.com/EBiCS/EBiCS_motor_FOC
- Display easy to customize, DIY and repair (hardware and firmware): https://opensourceebike.github.io/

[image]

[image]

https://img.youtube.com/vi/ueGZx3Urf0c/0.jpg