OpenSourceEBike/Bafang_M500_M600 #38: Using the modified firmwares

AnthonyMorley, 2022-01-16:
Hi people. Sorry if this is posted in the wrong place (but I am new to github and unfamiliar with protocols; this is the only place I could find to type a message).

I am wanting to try one of your modified firmwares on my 48v M600, using the BESST box.

Do I simply download a firmware from the firmwares page, and upload it to the bike via BESST ?

Thank you

Ornias1993, 2022-01-16:
We do not offer (working) modified firmwares at the moment.
THe other firmwares are just official firmwares.

closed 2022-01-16 by Ornias1993

AnthonyMorley, 2022-01-16:
Thank you for the immediate response.

I understand that they are experimental firmwares.

And I understand that there are no guarantees.

And I can see that you have worked very hard on this project.
And it is very kind of you to opensource the files and information.

I have official 14.7 M600 firmware installed on my bike.

Can I please ask what is the difference here from stock firmware  ...
- BIN -72.6 KB Firmwares/Custom/Motor/m600/experimental 48v20a.bin
- BIN -72.6 KB Firmwares/Custom/Motor/m600/experimental 48v25a.bin

Ornias1993, 2022-01-16:
As stated before we offer no working custom firmware.
So discussing that is kinda irrelevant.

Ornias1993, 2022-01-16:
Actually, I don't know what you are doing at all, becuase the files you referenced have been deleted from the Repo for a while now.
The discussion about it is public if you want to read into it.

TLDR:
There are NO (working) custom firmwares being offered by us at this time.