OpenSourceEBike/Bafang_M500_M600 #68: CRX10NC4814i101003.0-28-2230-boot3.0-20220218085143.bin for M510 controller cannot be flashed

briangdd, 2022-05-15:
CRX10NC4814i101003.0-28-2230-boot3.0-20220218085143.bin for M510 cannot be flashed.

File can transfer to BESST device, but no evidence that anyone has managed to activate the firmware transfer from the BESST device to the M510 controller.

The original firmware can be flashed with same equipment using the usual workflow:
1. Read info from controller using BESST
2. Disconnect BESST from M510 controller
3. Choose file and upload to BESST device
4. Plug in M510 controller to BESST device
5. Wait until transfer is complete and test light on BESST is lit

The aforementioned firmware cannot get past stage 4 under any circumstance.

Renaming file and shortening path have no effect.

DadoKomar, 2022-05-18:
Hello guys. 
What is new with new M510 firmware, which was uploaded recently. Should I understand it is still version 3.1 ?
Thx

KyokushinPL, 2022-05-18:
Users were unable to install the previous one. The new one have also improved battery attenuation.
Discussion about M510 firmware is here: https://www.emtbforums.com/community/threads/bafang-m510.26180/post-395217