OpenSourceEBike/Bafang_M500_M600 #81: New firmware M600 36V 18A

cpardo13, 2022-08-27:
Hello, 
I have a new firmware from Bafang for M600 36V 18A.
I have compare with the 25A firmware and have a little of bits of diferencies.
I think if a developer or expert compare this two files, know the bits to modify for personalize a firmware. 
Add a link to download the firmware because im not upload to this repository. 

https://we.tl/t-Er6noc3yAk

Ornias1993, 2022-08-27:
Everyone can open a PR to this repository, please do so.

Ornias1993, 2022-08-27:
Don't expect others to do that for you... We expect opening a fork, uploading an opening a PR.

closed 2022-08-27 by Ornias1993

cpardo13, 2022-08-27:
I don't understand sorry. My english is very bad.
I open this issue for share a new firmware because im not can upload directly.
But you're still as nice as ever.

Ornias1993, 2022-08-27:
Please file a PR (Pull Request), not an issue.
For your needless personal attack i'm going to lock this issue. It's not needed at all.