OpenSourceEBike/Bafang_M500_M600 #102: Cannot change speed limit on M500

miko671995, 2024-05-29:
Hello. Im using bypass login script. Everything seems to okey. Some problems with reading serial but sucessfully read.
After i want to write speed i got alarm that DEALER or Assembler are not allowed to increase it to bigger number. Can u help me with it ? 

Controller CRX10NC4813I142071.2B
CR X10N.250.FC
CR X10N250.FC 4.2

tried many versions of BESST software.

yearned, 2024-05-30:
I don’t know how to make it work ,sorry

miko671995, 2024-05-30:
can i help you somehow with debugging ? i never read over the forums that somebody was fighting with this issue.

ludwiglikar, 2024-06-06:
Sounds you would like to set the limit over 25km/h.
Version 1.2.18 works good to change speed limit up to 60km/h on all Systems.